rt-bishop/Look4Sat #108: Not updating

mebrown47, 2023-05-20:
Online update returns "update completed with errors"

DroidKali, 2023-05-20:
Same issue

sgcderek, 2023-05-20:
Transponders are updating, suggesting that the issue is the URL for the TLEs. Please consider adding back the option to set custom URLs

arctis-shark, 2023-05-20:
Same for me

cfd238, 2023-05-20:
Bummer,  same here.

mebrown47, 2023-05-20:
https://celestrak.org/NORAD/elements/gp.php?GROUP=active&FORMAT=tle

Use the Settings/File feature to load active.txt from the link above as a workaround.

rt-bishop, 2023-05-21:
Hello everyone! Thanks for reporting this and sorry for the late reply.
There are a number of reasons for this behaviour. 

Normally the update function would just skip the failing data providers and fetch the ones it can, so the message "Update completed with errors" actually means "partially completed". I know - not the best naming. But this time there is a problem with multiple data providers.

First, the McCants TLE data provider changed it's servers and TLE URLs, which were partially updated [here](https://github.com/rt-bishop/Look4Sat/pull/105), but I haven't created a new release build yet.

Second, our largest data provider, [Celestrak](https://celestrak.org/), has changed it's T&C, changed the domain and URLs, and introduced the excessive downloads IP blocking. So the app can't get access to the majority of data providers.

We still do have [AMSAT](https://amsat.org/tle/current/nasabare.txt) and [R4UAB](https://r4uab.ru/satonline.txt) and they seem to be working fine. There is also [Space-Track](https://www.space-track.org/auth/login) but it requires login in order to get the data and I'm not a big fan of introducing credentials into the app. 

So, as an ad-hoc workaround we can use the offline data update as @mebrown47 mentioned. Another fix that I can deliver short term is updating the TLE URLs for McCants and Celestrak, OR if Celestrak is going to block IP for excessive downloads - removing it and just sticking to McCants, AMSAT and R4UAB. We'll loose the majority of satellite categories (and some satellites) within the app but at least it'll continue to work.

[image]

sgcderek, 2023-05-21:
The skipping may not be working correctly, I installed L4S on a different phone, and it only updated transponders but got 0 satellites. 
Perhaps it would be possible to set up Github Actions or some other tool to periodically download the master tle file and host it here on github, say once a week, to reduce the amount of Celestrak requests

rt-bishop, 2023-05-21:
@sgcderek, yeah I can confirm that it currently only updates the transponders. I'll look into that.

Also, I heard that some projects store custom app configs and even serve certain APIs via GitHub, but I have a feeling it's at least in a grey zone if not totally inappropriate usage of this platform. It would mean that Look4Sat's ~30k users will just download the data from here instead of Celestrak. So I don't think it's a futureproof solution.

ckuethe, 2023-05-21:
I would love to see custom URLs come back; for past versions I wrote a proxy/cache thing to fetch TLEs from ... nonstandard sources.

Also, if celestrak is ratelimiting too much, check out http://tle.info/ ... this is another argument, I think, in favor of allowing custom URLs to come back. :)

rt-bishop, 2023-05-21:
@ckuethe Thanks a lot, as always! Yeah, now I see that custom URLs make sense. I'm working on a new version, but kinda slow, as I'm quite busy with my day job. But custom URLs will definitely come back when it's released. 

BTW, I've created a new build, it's currently being approved by Google but should get to the users shortly. It's working for me, but I want the rest of people to get it and test it before we know what the next step is.

jtfletchbot, 2023-05-21:
@rt-bishop I have downloaded the updated app and it fetches the new TLEs without error. However, #107 is still happening.
I even tried downloading TLEs manually and loading from file. GOES 16 doesn't show up by "GOES 16", "GOES-R", or by it's NORAD ID "41866". However, it does appear in Celestrak's database online.

It's not a major issues, I just find it odd that GOES 17 is present but not GOES 16.

closed 2023-06-03 by rt-bishop