rt-bishop/Look4Sat #109: Satelite name not displayed once selected

jreytinat, 2023-05-26:
Once a satelite is displayed, nothing shows which one it is, quite problematic when several are reachable, impossible to find out which one has been displayed previously. 
If possible, display in the satelite info its name.
Thanks regards.

f4iyy, 2023-06-25:
Hi,

Agree with @jreytinat , it will be very usefull in action phases.

Thx

lucasburlingham, 2025-02-02:
+1 vote for this feature. I may try to implement this myself but we'll see if the PR gets pushed.

rt-bishop, 2025-02-03:
Hey there! Just a quick update. This feature has been implemented, just like a few others as well, but is still waiting for the rest of the app rebuild to be done. I'm hoping to make a new release probably in a month or so.