rt-bishop/Look4Sat #112: RS-44 (DOSAAF-85) are not correct.

ON4ASG, 2023-07-05:
The frequencies in the App when tracking RS-44 (DOSAAF-85) are not correct.
RS-44 is an inverting transponder...
Uplink: 145935->145995 LSB
Down:  435670->435610 USB
Center: 435640
Beacon: 435605 CW

Look4sat App gives Mode V/U transponder: 435.6161-145.9329 which is impossible combination!
Would be great to have the possibility to enter your own custom downlink frequency and get the according uplink frequency with corrected doppler, see wish 4O4A.  :)

Grts,
Hans
ON4ASG

ON4ASG, 2023-07-06:
AO7, FO-29, CAS-4B, JO-97....all have the same issue

rt-bishop, 2023-07-07:
Hey! Thanks for reporting this)
Look4Sat reveices the data from the SatNOGS db in this format:
`
"uuid": "89NTmbqJEuTGW36fd386Fy",
        "description": "Mode V/U - Transponder",
        "alive": true,
        "uplink_low": 145935000,
        "uplink_high": 145995000,
        "downlink_low": 435610000,
        "downlink_high": 435670000,
        "mode": "USB",
        "uplink_mode": "LSB",
`
When it displays the frequencies it ignores the "high" values and only shows the "low" ones, calculating doppler for them.
It was easier to implement this way because most of the satellites don't have the "high" value and I needed to come up with something from the UI perspective. I'm happy to redo it, but there is always a balance between the amount of data and how to properly display it in the UI.
If showing all the freqs, then we have up to 4 fixed values and up to 4 doppler corrected ones. I can extend the transmitter cell but would like to know how to squeeze all the required data in the best possible way? Your suggestions (and drawings!!!) would be highly appreciated =)

ON4ASG, 2023-07-07:
Arty,Thanks for the quick reply.I do not write code so I can only explain our needs in human language :)First let me congratulate you with the Look4sat App, it is already one of the best tracking apps I found and tried to use.What I will try to explain is important for hamradio operators like me, who want to make contacts over the linear ssb transponders like RS44, AO7, FO29, etc...and using a half duplex system.(See also feature request #91 from April 9th 2022 by 4O4A).The FM birds who only have one channel like SO-50, ISS, Tevel-1 to 8 are not a problem. As they only have one FM channel it is easy to know where to expect the downlink after doppler.On 145Mhz doppler is never more than few KHz, on 435 MHz it can be up to 10KHz.Knowing that an FM channel is 12.5 KHz wide there is no need for doppler correction on 145, mostly the uplink.On the 435 downlink it is very easy to program 2 or 3 channels 5KHz away from the downlink and see where to receive.Also Look4Sat is doing a perfect doppler correction here.The problem starts with the SSB transponders using a narrow 2.7 KHz bandwidth in a downlink covering 60KHz ( RS44 = 435.610 to 435.670).If I start a CQ call on 145.965, I can expect an answer around 435.640 if there was no doppler.I can estimate what will be to doppler  by checking the difference with the beacon (435.605),Receiving the beacon 435.598 I can estimate a doppler of about 7 KHz.So when dialing around 435.633 +/- a couple KHz I can expect an answer on my CQ call.!!!The biggest issue is when I hear another station calling and I want to answer him. What uplink frequency should I use to be heard exactly on the frequency I heard the other station calling.I have to calculate the downlink doppler, then check the corresponding uplink frequency in the transponder, calculate uplink doppler....to much for my brain as I only have secondsI added 2 pictures helping explain the problem.So the one thing that could be a fantastic gadget is that one is able to fill in a downlink frequency and automatically see the doppler corrected uplink frequency!!!!!!!If one can realize this, it would be fantastic for the hamradio community, we will be glad to spend some money to buy the app.Feel free to ask further questions, glad if I can help you. Thanks for reading.Kind Regards 73,HansON4ASGVerzonden vanaf mijn Galaxy

dl3jop, 2023-09-09:
Hi, I may got a simple solution for it:
I wrote similar satellite tracking program which also parses the satnogs transponder data:
If there is a lower and upper value present I calculate the transponders center frequency as a preset. Using a slider, you can change the frequency. 
I guess adding a slider below the transponder frequency box to sweep the frequency between +-60 KHz from the center frequency could be an elegant solution which does not require a complete overhaul of the design?

ON4ASG, 2023-09-09:
Joshua,Than you for the effort and reply.Really looking forward to test it.Would be a huge improvement.Grts,HansVerzonden vanaf mijn Galaxy