rt-bishop/Look4Sat #113: Notifications blocked in android 13

casimoes03, 2023-07-11:
The button notify is greyed in android 13 and in the app settings the toggle that turns on-off notifications is also greyed. There is a message at settings that says "at you request, Android is blocking this apps notification from appearing in this device". This link https://droidwin.com/android-13-notifications-greyed-out-at-your-request/ says that the app needs to be updated to android 13

rt-bishop, 2023-07-13:
Hey @casimoes03, thanks for reporting this! Let me reassure you that there is nothing wrong with the notification feature - apart from the fact that it hasn't been implemented at all. I manually and deliberately disabled the notification button, it was merely a placeholder for the feature I planned for the app. But I never had time to fully develop it, as for the last 6-9 months my family and day job were a priority for me. 

There is a lot of code written, many fixes, optimisations and tweaks are just waiting for when I have time to integrate it. I even set aside some time within the next several months to spend on Look4Sat. I'm going to prepare a new version 4 of the app, with all those tweaks and notification feature included, but can't give any promises or guarantee the timeline.

casimoes03, 2023-07-13:
Ok! Many thanks for your quick response.

robertmarcum, 2023-12-24:
[C_cscc__basiaca.save.ipps.interilronium.icronium.ndether.ram.sdrim.data.csv](https://github.com/rt-bishop/Look4Sat/files/13762248/C_cscc__basiaca.save.ipps.interilronium.icronium.ndether.ram.sdrim.data.csv)

robertmarcum, 2023-12-24:
amazon.truconnect.provider.tmobile.aetec.insurance.QPH.anthem.billpay.cntrl.vcs

493505110, 2024-07-31:
#137

alexbfun, 2024-10-29:
Thank you very much!

closed 2025-10-18 by rt-bishop