rt-bishop/Look4Sat #114: App doesn't seem to point right direction

2fst4u, 2023-09-04:
When I try to find a sat, it tells me to point at a downward angle for a satellite that I know is above me. Am I doing something wrong or misinterpreting? For a satellite with 60 degree elevation I'm looking at my screen in my hand with a natural downwards angle as.if you'd normally look at your phone to read a text.

My azimuth is also slightly out when comparing to other apps on my phone.

Is there something I've done wrong? I've updated location and data several times and even moved country.

bobboau, 2023-09-05:
I was trying to figure this out too, was expecting the target icon to represent the satellite and if I get it centered on screen was expecting that to indicate my phone was pointing at it, but the way it behaves I have no idea what it's supposed to be doing or what any of the graphics mean. What's the orange dot? Am I correct about the target? I get the red li e is supposed to be the path, but shouldn't that mean the target would be somewhere on that line? Maybe there's just some bug making this screen misbehave?

2fst4u, 2023-09-05:
Yea agree I just can't figure out which way I'm supposed to be looking.

rt-bishop, 2023-09-05:
Hey everyone, thanks a lot for reporting this! I think I may need to integrate some sort of guide into the app to clarify those bits. The whole moving radar feature was created just so the phone can be attached to the antenna to make easy radio contacts via a satellite with a handheld radio. 

So the yellow dot on the radar screen means the current satellite position, the red line is the satellite's track when it passes overhead. The white cross means zenith (90deg sign gives a hint to the scale), the external white circle is the azimuth (0-360deg). So to sum this all up, the current satellite's point is not seen like "through the camera" or like "you read the text on the phone" (Red Z axix), but rather is the continuation of the device's Y (blue) axis pointing to the place in the sky.

[image]

This way attaching the phone to the antenna seems natural as you can follow the satellite's track by hand making both low and high elevation QSOs.

I double checked the calculations and everything seems to be working fine. I mainly use GPredict on desktop and it draws the same Radar view and the same satellite track as Look4Sat for any location I set. 
Although it may seem unnatural that satellites are so low, but actually most of the time they are relatively low with <45deg for the highest point in the sky. Getting a really good high evelation pass is not always a thing and for some satellites may happen just a few times a day. This heavily depends on the type os satellite, it's orbit and the time of day.

Hope this info makes the app a bit easier to use. I'll think about how to fit this info into the app.

bobboau, 2023-09-05:
OK, so if I get the red target over the yellow dot the top of my phone will be pointed at the satellite? Might be good to have an option to use the (negative) z axis as the pointer, that seems more intuitive if you're just trying to see it.

2fst4u, 2023-09-05:
I had a feeling that might be the case but the lack of indication or explanation did make it hard.

Can I suggest a toggle to change it from the current method to the method we were anticipating it working as? That is, pointing your camera at the satellite? 

The method you describe is good if aligning a feedhorn, but if I'm scoping out a site for satellite shots without the actual hardware on me, I want to use the point-camera-at-sky method so I can visualise the direction and if there are any obstacles.

There is also an alternate use-case that you might want to align a back edge of a dish that is known to be 90° vertical compared to the feedhorn. I have used analogue measuring tools in the past for this like an inclinometer attached to the rear.

I'm hoping a toggle like this would be a fairly simple addition?

rt-bishop, 2023-09-05:
Yes, precisely the red aim should be over the yellow dot in order to point straight at the satellite. And the yellow dot is only displayed if the satellite is currently above ground.

The use cases you mentioned are actually something I completely overlooked. Yes, it definitely makes sense to add an option for that. I'll add it in the next release, but taking in account the amount of free time I have it may happen somewhere ~end of October.

bobboau, 2023-09-05:
Might also be a good idea to orient the rotation of the radar by the compass rather than the accelerometer. 

A link to the radar from the info summary in the map (that you get when you tap on a satellite in the map) would be cool too.

rt-bishop, 2024-05-08:
Hey there! The radar feature was implemented to cover my use case obviously, which was putting it on the Arrow antenna for VHF/UHF contacts. However there are plenty of use cases which this implementation does not cover. I should definitely integrate some UI indications, scale and some sort of an in-app guide to explain how it works. For now just as mentioned above the top of the phone points to the satellite when the red target aim is placed over the yellow dot when the satellite is above ground.

The radar UI was mostly taken from the GPredict desktop app, so I was not putting too many thoughts in it. Also I recently became a daddy so currently there is no time to continue the app's development. However I do have intentions to do a complete UI overhaul with an in-app guide, BottomNav, blackjack and all that jazz and even got a half ready build. Once I manage to get my life/work/hobby balance back together I'll make sure to integrate most of the planned features.

umitseyhan75, 2024-05-09:
I was trying to use the app to position the satellite dish (and LNB) to find the perfect angle (azimuth, polarization, elevation) for the television. So it is not an arrow antenna. I am aware that the top of the phone points to the satellite, but what I am not so sure is that the "face" of the LNB is looking at the dish and not to the sky, so I am confused that whether should I aim from the LNB collar or from the center of the dish?