rt-bishop/Look4Sat #115: GPS constellation: display Pseudo-Random Noise (PRN) number

rouseabout, 2023-09-04:
Feature request: for GPS satellites, display PRN number on the transceivers data page.

All NAVSTAR/GPS satellites transmit on the same L1/L2 frequencies, and use Pseudo-Random Noise (PRN) sequences, or Gold codes,  to differentiate themselves from one another. PRNs are published here: https://www.navcen.uscg.gov/gps-constellation Another list with NORAD IDs: https://www.n2yo.com/satellites/?c=20

Alternatively, Issue https://github.com/rt-bishop/Look4Sat/issues/78 ("own short comment to each tracked satellite") might satisfy this feature request.

rt-bishop, 2023-09-05:
Hey @rouseabout, thanks for reporting this. Look4Sat uses various providers of TLE files, which only contain the info for the satellites orbits, and uses SatNOGS api to get info about the active transponders on the satellites. Since PRN info is only related to transponders, it should be received from transponders data providers. 

Ideally it would be great to get SatNOGS to display GPS transponders with frequencies and all the other info you may need just like they do for Amateur satellites. https://db.satnogs.org/api/transmitters/

So it'd be great if you're willing to chat to SatNOGS about including this additional data into their DB and API. It'd be even better to cover much more transponders, like the Starlinks for example. I'm happy to extend the amount of fields I read from their api, to make the UI adjustments for additional info regarding the GPS transponders, but I don't see much value in parsing separate websites just to get this piece of info.

Alternatively, you may want to wait until I implement #78, and I can't give you any estimations on that.

rouseabout, 2023-09-06:
Understood. Thanks!

closed 2023-09-06 by rouseabout