rt-bishop/Look4Sat #116: Is iOS version possible?

liups233, 2023-10-09:
I am a new HAM and wanted to chat over satellites. Look4Sat is the best satellite tracking software on mobile devices which can help me aim the antenna at the satellite and check the frequency. But on iOS, I did not find anything convenient. Is there any plan to put Look4Sat on Apple devices?

rt-bishop, 2023-10-11:
Hey @liups233, thanks so much for your question! 

It certainly is possible, the app is open-source, although I'm not sure who will do it =) Generally I prefer to stay away from Apple tech because of their policies like "🍏way or no way", having to own a Mac to develop for iOS etc. Also to ship it via AppStore there is an annual licence fee for developer's account. Some would say "Ooh, it's only 100 bucks", but that means either 100 bucks out of the pocket, or to make this app no longer free, which I don't want to do.

However, I'm currently rewriting this app in Jetpack Compose and with KMM in mind. There is some work done by JetBrains to run Kotlin Multiplatform and Compose apps on iOS, so technically it'll be much easier to ship a similar codebase to both platforms. Although the aforementioned problems still arise. So I don't have a proper answer to your question at the moment.

liups233, 2023-10-11:
Got it. Anyway, thanks so much for developing such a good software to help us!