rt-bishop/Look4Sat #117: explain how to use the app to beginners.

Abdull, 2023-10-28:
this is for v. 3.1.3.

I have no idea how to use the app.

My initial intent for exploring this app was to find satellites visible to the naked eyes upcoming at my current location (a filtering / sorting by "apparent magnitude of satellite (reflectance of sun light) " does not exist, right?).

But I got confused with several things during the app usage, which are not clear to me as a beginner. 

I could imagine a guided intro tour or explanatory info bubbles could help with onboarding. 

Here are a couple of annotated screenshots.

[image]
[image]
[image]
[image]
[image]

rt-bishop, 2023-10-31:
Hello @Abdull! I'm super impressed with your thorough approach, and will happily answer all these questions. 

Initially the app was created for amateur radio operators to track Radio satellites. The app even had "Radio" as a part of it's full name in Google Play. The idea with the name was that "look for" means searching rather than actual looking at the satellite. Naming is in general a really complicated matter =)

You're right - there is currently no such thing as satellite's magnitude in the app and in general the UI is extremely cryptic for new users =) Look4Sat is my side project but unfortunately because of the main job and personal issues I don't have as much time for it as I used to have before. I do have quite a lot of ideas about how to fix the UI, make it more self-explanatory to new users and even planning to integrate the guide with FAQ into the app itself. I do have some annual leave coming up and will try to use that time to make progress.

As for the questions, let's start by adding some context:

### -1. General notes
The app is never meant to be used while selecting ALL the satellites. Currently the database has ~9131 satellites and thanks to Starlink you can easily have hundreds of satellites above your head at every single moment. There is no way (and no reason) this can be displayed in the UI, that's why satellite categories are meant to be used. Let's say one wants to observe Starlink newly launched "trains", this can be easily done by unselecting all the satellites, and selecting all from the Starlink category. The same story with Radio satellites or ISS. It's crucial to know what you're looking for and select only that otherwise it's just too much distraction.

### 0. Navigation
The app has several screens:
- Satellite selection screen, with ticks and categories.
- Passes overview. The page where you land initially.
- Single-Sat "Radar" view is shown after you click on a pass.
- Multi-Sat "Map" view is shown after you click on a planet icon.
- Settings screen which contains the general setup tweaks.

### 0. The Countdown
This countdown shows the time until the next active pass. When the satellite raises above horizon observed from you location, it's pass becomes active and the yellow progress indicator shows the current status of the pass. Once the satellite sets below the horizon, its pass gets deleted from the list.

### 1. Passes
According to SGP4 model, satellites can be NearEarth (rotational period < 225 min) or DeepSpace (rotational period >= 225 min).

Most of the satellites fall into the first category and they make one rotation every several hours. Their passes take ~10-15 min and the progress bar will indicate their state. You'll also see the time they rise above the horizon (AOS - acquisition of signal/sight) and set below (LOS - loss of signal/sight).

Other satellites (like Trumpet and Polar on your screenshot) will be in the DeepSpace category and will either never set below the horizon, or they can be above for hours. They won't have AOS/LOS time and their progress bar will always be fully yellow.

### 2. ISS
They represent different modules of the International Space Station. The data comes from various providers, the naming can be quite different between let's say the AMSAT data and the Celestrak one. Sometimes you have to try several names to find what you're looking for. The easiest way is to search by the [Catalog Number](https://en.wikipedia.org/wiki/Satellite_Catalog_Number) - it's always the same, for ISS it's 25544. Check this out - even the Catalog Number itself has 4 different names =)

### 3. Radar
The aim on the radar shows the place in the sky you're pointing with your phone. The satellite above ground will be shown as a yellow dot on the red trajectory. The phone can be easily attached to an antenna and one can track the satellite movement by hand while making a radio contact (QSO) with another amateur radio operator.

### 4. Eclipsed (or Visible)
This means that the satellite is currently not in sunlight, eclipsed by Earth. This is crucial to know as satellites normally switch off the transponders in that case to save energy, and it's not possible to use them for QSO. Once they go back into sunlight, their solar panels produce enough energy for transponders to function.

### 5. Passes filter
Passes are not recalculated every second. You can only calculate them for a certain amount of hours ahead. If you set 2, this means you want to see the passes for the satellites you selected that will occur within the next 2 hours. If you set 0 it will show you the passes within the next hour, but some satellites still haven't rised above ground, so the countdown shows the next AOS.

### 6. Map view
Map doesn't deal with passes, it shows only the current positions of all the selected satellites, and the radio range with trajectory for the currently selected one. Again, there is absolutely no sense trying to see all ~9k satellites on the map. They simply cover all possible space on the map and cripple even the modern smartphones. 

And yes, it's exactly that flooded above our heads and they keep sending more and more sats there. Luckily the Space is called that way to describe that there is enough space for everyone =) Let's see how many years it'll last.

### 7. The end
Hope this brings at least some light into how to use the app. In the next app version I'll try to address the issue of a very poor and unclear UI, will try to add the info about the satellites magnitude and implement some things that I planned for a long time. Without the family matters and the day job it used to be much easier to support an open-sorce app =)

juanpc2018, 2023-11-28:
try [TrackSat](https://apkpure.com/tracksat-satellite-tracker/hr.from.robertpetrovic.tracksat)
= 3D Globe

closed 2023-11-30 by rt-bishop

Abdull, 2023-11-30:
(@rt-bishop, I haven't had the time to respond in time) .

Thank you very much for your explanations; they made the concepts clear to me.

In case you see fit, incorporating such explanations into the app would IMHO help clueless newbies such as me with onboarding.