rt-bishop/Look4Sat #119: satellite display wrong cover range in map site

EeeLo, 2023-11-22:
I'm love and using old version by Look4sat, but sometimes some satellite display wrong satellite cover range in map site,any good idea?BTW,I want keep old version,thanks for your good project.[image]

rt-bishop, 2023-11-29:
Hello @EeeLo, sorry for such a late reply. Indeed this is a known bug and its reason was hidden deeply in how OSMDroid library (which shows the map) handles and optimizes the area path internally. The fact that the map has infinite horizontal scroll slightly messes up the path coordinates. I could disable it but eventually found another solution, which however, exists only in the latest version of the app.

I'm glad to hear you like the app, but you mentioned you use an old version. Can I ask, is it because of the interface and the Bottom Navigation? If yes, you'll be happy to know that I'm preparing a new version with that navigation coming back. It proved to be much simpler and much more straightforward. If everything goes well I'll deliver it some time in January.

EeeLo, 2023-11-29:
Thanks for your message,I like this app very much because it's so easy and friendly to use,I like the Bottom Nav in old version,can you tell me how to fix it at this time?Can't wait to use feture version and thanks for your great project!

EeeLo, 2024-02-28:
> 你好@EeeLo，抱歉这么晚回复。事实上，这是一个已知的错误，其原因深深隐藏在 OSMDroid 库（显示地图）内部处理和优化区域路径的方式中。地图具有无限水平滚动的事实稍微扰乱了路径坐标。我可以禁用它，但最终找到了另一个解决方案，但是该解决方案仅存在于该应用程序的最新版本中。
> 
> 我很高兴听到您喜欢该应用程序，但您提到您使用的是旧版本。请问，是因为界面和底部导航的原因吗？如果是，您会很高兴知道我正在准备一个新版本，该导航将会回来。事实证明，它要简单得多、直接得多。如果一切顺利的话我会在一月份的某个时候交付。

Hi friend, can you tell me how to fix this bug on old version? I can fix it for use earlyer.