rt-bishop/Look4Sat #120: where is the "File" ?

juanpc2018, 2023-11-28:
[Online] method, does Not work,
Offline [File] method, but where is the file?

rt-bishop, 2023-11-29:
The app uses TLE data that usually comes in .txt files. And you're right the data is mostly grabbed from [Celestrak](https://celestrak.org/NORAD/elements/), just select TLE/3LE and then any group like [active](https://celestrak.org/NORAD/elements/gp.php?GROUP=active&FORMAT=tle) which includes them all. There are other providers like [AMSAT](https://www.amsat.org/tle/current/nasabare.txt) or [McCants](https://www.mmccants.org/tles/) that has classified satellites in a [zip](https://www.mmccants.org/tles/classfd.zip), just make sure to unzip it first.

Also what issues exactly are you experiencing with the online update? I've just checked, it works as expected. Some time ago Celestrak moved to a new domain which caused issues with the older app versions, but the most recent one v3.1.3 works as expected. Anyway the offline update should be working just fine.

juanpc2018, 2023-11-30:
> Also what issues exactly are you experiencing with the online update? I've just checked, it works as expected. 

https://github.com/paveloom-f/lbry-android/issues/6