rt-bishop/Look4Sat #126: Satellite predictions are in correct after Feb 29th

t0nitogp, 2024-03-02:
I believe there's a leap year issue, after Feb, 29th all the predictions are incorrect, like if it's predictions were from a different location, for example, GOES 16 from my location is 8.8° elevation is now showing as 9.5°. The prediction times are incorrect as well as the paths. Opening a satellite to see it's current location is also different from the other prediction software/sites.

RAD750, 2024-03-02:
I can confirm, but also there is a workaround
Clear all the satellites, then download the Active catalog GP from Celestrak, and import it to Look4sat.
Link for the file is here: https://celestrak.org/NORAD/elements/gp.php?GROUP=active&FORMAT=tle

Remember to disable autoupdate

t0nitogp, 2024-03-02:
Great! Thanks it worked! I hope they fix the issue soon though 👍

rt-bishop, 2024-03-02:
Thank guys for reporting this so quickly. So I've checked and the calculations that are made are correct. As you already figured it out yourself the issue seems to be in the TLE data that hasn't been updated. Look4Sat uses several data providers apart from Celestrak. And their data seems to be out of date. After satellites are imported they are stored in the internal DB, with CatNum used as their unique key. Celestrak data can get ovewritten with other providers' data after a regular automated update.

So it seems there is nothing to fix as such. Once every provider updates their data and then in turn users update their data, the issue should go away by itself. Interesting quirk that I need to keep an eye on in the future. Maybe I'll manage to alter the calculation so this issue never comes back. Or maybe I can implement the data provider selection while retaining the search ability. If only data formats and extensions were standardized...

t0nitogp, 2024-03-02:
If I may use this to make a request, is it possible to have an option to use look4sat in landscape mode? I use a portable setup using my phone for HRPT reception, having the phone sideways, it would be nice to be able to use look4sat in this landscape mode.

rt-bishop, 2024-03-03:
@t0nitogp, of course, it's an amazing request. I had this in mind for quite some time already. I'm in the middle of converting the app's UI from the old XML view system into Jetpack Compose. A lot of stuff gets rewritten and the whole UI concept needs a slight revamp. So I'm definitely working on it, but I can't give you any estimates or timelines.

t0nitogp, 2024-03-04:
That's great news! Thanks!

9a6ww, 2024-03-04:
It's fine that you use many TLE sources, but maybe you can change the loading order and give priority to the most reliable sites like celestrak.org. I am sure that this will minimize the chances of such mistakes in the future.

t0nitogp, 2024-03-09:
Look4sat still isn't predicting correctly while other prediction software is. I think this really should be addressed.

rt-bishop, 2024-03-09:
I'm planning on excluding the data providers with stale data and leaving only the reliable ones like Celestrak, Amsat etc. This should rectify the issue but will also slightly affect the number of satellite categories the app has. I'm hoping to make a release today with this issue sorted.

t0nitogp, 2024-03-09:
That's great, will it still be able to display transceiver data? As manually updating the files from Celestrak won't display them.

rt-bishop, 2024-03-09:
Yes, of course, the transmitter data will be unaffected by this tweak.
I'll also add a hardcoded filter for now to only display the active transceivers, instead of all of them, even the dead ones.

rt-bishop, 2024-03-09:
The fix is now out and is being reviewed by Google. Should be available for download from GP in a few hours and from F-Droid probably tomorrow, depending on when their build chain is triggered. Sorry, I should've looked into this issue earlier. Make sure to do another data update after the app is updated.

t0nitogp, 2024-03-09:
Excellent! Thank you so much! This is the best satellite prediction software for Android, keep up the good work!

KJ7LNW, 2024-03-09:
Maybe add a simple UI with a list of URLs that can be toggled?  This could enable custom URLs, too.

erebion, 2024-03-14:
> Sorry, I should've looked into this issue earlier. Make sure to do another data update after the app is updated.

Are you getting paid for this? If no, don't be sorry and thank you for what you do.

frozenstein, 2024-04-02:
Hello. It still shows incorrect data for me. I have version 3.1.4, the latest that PlayStore offers me. However, I have found out one thing -- GPredict shows a RS-44 satellite pass tonight, that exactly matches to the incorrectly predicted pass I attempted yesterday, based on Look4Sat... Couldn't there just be a problem with 29th February? It seems to be just shifted by one day.

rt-bishop, 2024-04-03:
Hey @frozenstein! Cool nickname you have =) Thanks for reporting this. I've just compared Look4Sat with the latest data, GPredict and N2YO. All the sources show the exact same information in terms of satellite positions and future passes. I verified several other sats and the results are the same. So I can't really confirm that.
Yeah, the Feb 29th messed some things up, but by now all the data and calculations should be accurate.

I'd probably also play with passes filters, like setting the min elevation, to include or exclude certain passes.

closed 2024-05-05 by rt-bishop