rt-bishop/Look4Sat #127: Feature: Show date on the satellite list

KJ7LNW, 2024-03-06:
Hi there, I love your app!  Hopefully this is a simple change:

We like to do 10-day predictions, so having a date beyond 7 days instead of just a day of week would be useful.

This would help with scheduling calls for son Zeke KJ7NLL's [ISS contact project](https://youtu.be/NMwvoyL542Y).

Thanks!

-Eric, KJ7LNW

rt-bishop, 2024-03-09:
Hi there @KJ7LNW. Such an epic project, it looks so incredible! Glad that nowaday kids are still interested in SatCom and Radio... I've added the date display to each satellite pass in the recent update. It's currently being reviewed by Google.

Also it's good that you mentioned 10-day predictions. I had an assumption that nobody needs such a long term prediction and was about to reduce it even more, replacing the hours box with a slider... Maybe I should switch hours for days then to keep the UI simple. It's just I'm half way through a UI revamp and it's important to know what features are being used, what are not.

KJ7LNW, 2024-03-09:
Thanks for adding/updating the feature! When does it hit F-Droid?

Recently we wanted to see predictions out to March 22nd when the ISS turns on its packet radio.  That is far enough out that the TLEs might not be 100% accurate, but it would give us an estimate within probably 10-30 minutes.

Zeke is having a great time with satellite work, I would say it is his passion.  He keeps a map with pins of his many ground contacts through the ISS repeater.  His latest project is to build a [desktop satellite tracker](https://youtu.be/RH70UJO9QC0) as an educational tool for teachers.

rt-bishop, 2024-04-07:
You must be such a proud dad! These are some impressive projects, incredible effort and so cool it's on YouTube. You're both explaining things in a really simple and comprehensive way, such a great and useful content! 

When I was a kid I could only dream about having access to all that tech... I'm also becoming a daddy very soon and I have a feeling my hobbies will go on pause for a bit. But once things stabilize I'll be able to finish the new Look4Sat version with the nicer UI and some optimizations. 

Of course the current v3.1.4 is already on F-Droid, it usually takes about a day to reach it once the app is released on GP. But of course you already know that, it's just me finding time to reply on GitHub only a month after the question is asked, ha-ha.

closed 2024-05-05 by rt-bishop