rt-bishop/Look4Sat #128: IO-117 (greencube) next pass not listed when current pass is in progress

kconkas, 2024-03-09:
I noticed that when there is a Greencube (IO-117) pass in progress its next pass isn't listed in Look4Sat. Hitting the refresh button doesn't help. I am certain about the next pass' existence because several other prediction tools, such as Gpredict or Amsat.org online prediction tool, whose predictions usually agree with those of Look4Sat, 
all list them. LEO hamradio satellites don't seem to be affected by this issue.

kconkas, 2024-03-09:
Update: it isn't just the next pass while one is visible. It appears that the 2nd next pass is  affected.

rt-bishop, 2024-04-07:
Hey @kconkas! Sorry for such a late reply. There was a data related issue #126 that was resolved in the latest app version v3.1.4. If you still observe the same issue, I'd suggest to play with the passes filter dialog in the bottom left of the passes page.

The precision of passes calculation is +/- 5 sec and that slightly affects all the other pass data like the max elevation for example. It is possible that it just goes below and above your set filter and dissappears from the list. Maybe one day I'll go through the pass calculation process and set it to use monotonic time, that should improve the precision and fix this once and for all.