rt-bishop/Look4Sat #129: ESP32 Bluetooth transmission not receive

Cyfarw9dd, 2024-04-04:
Hi, I am trying to use the Bluetooth function to control my rotator. I used ESP32 to receive the data. I am certain that I input my device address correctly, also with the default data format. To exclude the problem, I used a debug app to connect my ESP32, and try to transmit some single numbers or dozen of them. They were all received, I can check them in my serial port. After I checked my code, I wondered there is any problem with the transmission function of the application. 
Anyway, eagerly await your answer, also thanks a lot for your marvekous software.

rt-bishop, 2024-04-07:
Hey @Cyfarw9dd! I'm really happy you're enjoying the app =) Thanks for reporting this. I only just now got some time to properly respond to this issue. 

Initially it was ajohns1288 who implemented the data transfer over Bluetooth https://github.com/rt-bishop/Look4Sat/pull/87. I never used this feature myself and don't really have any hardware to test it on properly. All I know is that it was covering his and some other users needs. 

I also heard that for some users only the debug version of the app sends the data correctly. Is that what you meant when you said "I used a debug app to connect my ESP32"? Was it Look4Sat's debug version build via Android Studio? If yes, then I assume that it may have something to do with the app minification step that happens on the release build. R8 minifier removes and obfuscates some code and maybe it also removes stuff needed for BT to work properly. Normally it's resolved by adding certaing rules to minifier config, but I won't be able to verify that. 

Or can I use another PC to simulate your workflow with things like ncat or anything?

I'll look into that and leave the issue open. Although I'm not sure when it's going to happen. Currently I have an unfinished version of Look4Sat rewritten in Compose with nicer UI, bottom navigation and other nice additions. But also I'm becoming a daddy in a couple of weeks, so there are plenty of other priorities that take place and in the foreseeable future. Once things stabilize I'll be able to finish the new version and keep the minifier issue in mind.

Cyfarw9dd, 2024-04-07:
Thanks for your response, I also rejoice to read your comment!

Regarding the debug software, I did think of to use the Android Studio to debug with, though I'm not good at application development. I turned to the application store and downloaded a app with the Bluetoorh serial port debug function. I use this app to transmitt data to my device and received successfully. The device also logged the connection to the app, data reception, and disconnection. However, When I used Look4sat and turned on the Bluetooth function, the device's log only showed the initializing part, with nothing regarding the connection and reception.

I believe the BT rfcomm data will transmitt to SPP device correctly. Perhaps something went wrong in this progress, I may check  Look4sat's log to vertify if the data is being sent properly. Or I can use the debug version of the Android Stduio through your method, to vertify. 

Anyway, for now, I will use the other feature to instead of BT for now. I'm still a fresh learner in this case, but once I make any progress in BT, I will update the issue(after doing the rotator XD). Also, best wishes to you. I wish you both all the happiness and joy as you prepare to welcome your little one into the world!

Cyfarw9dd, 2024-04-07:
Thanks for your response, I'm also glad to read your comment! 

Regarding the debug software,  I considered using Android Studio for debugging, although I'm not good at application development. I turned to the application store and downloaded a app with the Bluetoorh serial port debug function. I use this app to transmitt data to my device and received it successfully. The device also logged the connection to the app, data reception, and disconnection. However, when I used Look4sat and turned on the Bluetooth function, the log only showed the initialization part, with nothing regarding the connection and reception.

 I believe the BT RFCOMM data should transmitt to the SPP device correctly. Perhaps something went wrong in this progress, I may check up Look4sat's log to vertify if the data is being sent properly. Alternatively, I can use the debug version of Android Stduio, following your method, to vertify.

 Anyway, I will use the other feature to instead of BT for now. I'm still a fresh learner in this case, but once I make any progress in BT, I will update the issue(after doing the rotator XD). Also, best wishes to you, wishing you both all the happiness and joy as you prepare to welcome your little one into the world!

F4HTB, 2024-08-08:
Hello dear OMs,

I just took my project out of storage.
I’m also using an ESP32.
As you can see here: https://github.com/rt-bishop/Look4Sat/issues/104, I encountered issues as well and indeed had to compile version 3.1.2 myself. This worked but not through the Play Store version.
Now it works with the new version 3.1.4 from the Play Store.
Also, be aware that if you're looking for a carriage return at the end of the data transmission, you need to add \n at the end in the settings.
In my case, I format it like this:

AZ$AZ EL$EL\n

Here’s my repo for the code:
https://github.com/F4HTB/esp32Rotor

73!

Cyfarw9dd, 2024-08-09:
Hi F4HTB，
Thanks for your response. Seems like my problem is because not adding the \n, and I was working on the version 3.14 either. Since this problem occurred, I've turned to use the TCP/IP to transmit the data, and now is working very well. Only thing I need to solve is the stepper motor coding.
Thanks again for your comment,
73!

closed 2025-10-18 by rt-bishop