rt-bishop/Look4Sat #130: Incorrect NOAA prediction

arkangel-dev, 2024-04-05:
The app is reporting an invalid pass. It says there will be a NOAA 15 pass today (5th April 2024) at 18:40 but NOAA 15 is going to be over the atlantic ocean at that time according to both Orbitron AND the apps map. I am located in the Maldives so it shouldn't be predicting that

https://github.com/rt-bishop/Look4Sat/assets/31138457/1e490c13-7b69-4496-ad08-4f5e7ad9e667

arkangel-dev, 2024-04-05:
Wiping the apps storage and cache fixed. Will report again if it happens again

scarlettekk, 2024-04-06:
https://github.com/rt-bishop/Look4Sat/issues/126

rt-bishop, 2024-04-07:
Hey there! I've double and triple checked all the information and can't confirm your observations. I've tested Look4Sat from Google Play with fresh data, GPredict, Orbitron (I thought Look4Sat's UI is not ideal, lol) and N2YO. All the sources show the same data for NOAA 15, 18, 19 and several other satellites. 

Your issue does seem to be connected to #126, but that issue is resolved now and all the calculations are correct. I only left it open for some time to gather people's feedback. I'd suggest to do a clean app install instead of just wiping cache. Apart from that there is not much that can be done, as it shows correct positions and passes on all my devices. And I get confirmations that the app works correctly in several user chats.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

mklemmingen, 2024-04-09:
I can confirm that the bug exists. I succesfully use it on an android smartphone, but doing the same on a samsung tab yields the same bug as of arkangel. All settings are the exact same, and wiping both app data, as well as on another occasion the satellite data did not fix the bug. Reinstalling did also not fix it.

mklemmingen, 2024-04-09:
Context: 

on the tablet:
[image]

on the smartphone:
[image]

scarlettekk, 2024-04-09:
Is the tablet maybe running an old version of the app with the leap day bug sources in it?

mklemmingen, 2024-04-09:
Good point, I checked, and it does run 3.1.3 instead of 3.1.4. My repos were not up to date. Thank you for the quick resolve. Much appreciated

scarlettekk, 2024-04-09:
@arkangel-dev can you also ensure you are on version 3.1.4?

arkangel-dev, 2024-04-12:
sorry, I was on vacation. Yes it is on 3.1.4 @scarlettekk

arkangel-dev, 2024-04-12:
I'll let you know if this happens again

closed 2024-05-05 by rt-bishop