rt-bishop/Look4Sat #131: Hi 3.1.4 Unable to update transceiver

alyosha0991, 2024-05-12:
3.1.4 Unable to update transceiver
Display: 0

rt-bishop, 2024-05-12:
Hey @alyosha0991! Thanks for reporting this. However I can't confirm this issue, for me it works fine on all the devices. Could you check if this link - https://db.satnogs.org/api/transmitters/ opens for you? It may be your ISP blocking certain addresses or some DNS related issue.

ntoulasd, 2024-09-27:
I have the same problem.
The problem is with android <7.1 and Lets Encrypt certificate of db.satnogs.org.

*To developer, could you edit to use user certificate always, not only in debug at file app/src/main/res/xml/network_security_config.xml

like this 

<?xml version="1.0" encoding="utf-8"?>
<network-security-config>
    <base-config>
        <trust-anchors>
            <certificates src="user"/>
            <certificates src="system"/>
        </trust-anchors>
    </base-config>
</network-security-config> 

*To user
See more documentation and to add certificate as user in your phone here https://letsencrypt.org/2020/12/21/extending-android-compatibility.html

InstantRain, 2024-10-30:
i’ve met the same problem(