rt-bishop/Look4Sat #138: Not able to control rotator with Look4sat

t0nitogp, 2024-08-27:
I used to be able to control my wifi rotator by running rotctld on my PC and connecting the phone, the PC and the rotator. In other words by relaying the look4sat through the PC to the rotator on an older version of look4sat. Now I get errors on my rotctld window like "command '2' not found" or "command '5' not found"

[image]

t0nitogp, 2024-09-21:
Anybody?

rt-bishop, 2024-09-22:
Hey @t0nitogp, sorry for such a late reply. I've recently became a daddy so for the last few months wasn't paying attention to this project at all. But I think I know what your issue may be.

In the previous version I've released a change to the rotctl command the app is sending. One of the users said his device is not receiving the long commands (see https://manpages.ubuntu.com/manpages/trusty/man1/rotctl.1.html#commands) so I've switched to the short command ("P $azimuth $elevation" instead of "set_pos $azimuth $elevation"). Also short commands are used by the GPredict desktop app, so more alignment on that.

I've never worked with rotctld on Windows but I expect it should have a configuration option to force it to read the short commands if it's using long by default. Alternatively, I can just make the change in the code, build an APK and send it to you, as currently I'm not in a position to make another release. Also if some users expect short commands and others expect long ones, it's going to break things. Wish those daemons were handling that by default.

Let me know if that helps at all and if you want me to send you an amended APK.

t0nitogp, 2024-09-22:
Congratulations :)
I would like to have the apk if you don't mind, that's really kind of you.
Thank you very much.

rt-bishop, 2024-09-22:
Thanks so much! It's been a roller coaster =)

[Here](https://drive.google.com/file/d/1a6wYmUMXlWFWZblOPV_QUajLU8GFF9MC/view?usp=sharing)'s the link with the reverted change to the data reporter. Hopefully that fixes things on your side. Once I get to another app release I'll review those rotctl commands once again.

t0nitogp, 2024-09-22:
Thank you very much 👍

closed 2024-09-22 by t0nitogp