rt-bishop/Look4Sat #140: How to Retrieve the Frequency of a Specific Transceiver in Look4Sat via Bluetooth Data?

shenzihang, 2024-09-04:
#### Issue Description:
I am currently working with Look4Sat and need to retrieve the frequency of a specific transceiver through Bluetooth data. While I have successfully obtained the azimuth data, I am uncertain about how to extract the frequency information. 

#### Steps to Reproduce:
1. I examined the relevant code and found the following snippet:
   ```kotlin
   var buffer = format.replace("\$AZ", azimuthString)
   buffer = buffer.replace("\$EL", elevationString)
   buffer = buffer.replace("\\r", crChar.toString())
   buffer = buffer.replace("\\n", nlChar.toString())
   buffer = buffer.replace("\\t", tbChar.toString())
   ```

2. This code allows the addition of azimuth, elevation, and control characters into the Bluetooth data buffer.

#### Question:
- How can I modify or extend this code to include the frequency of the transceiver in the Bluetooth data?
- Is there any relevant documentation or reference that explains how to achieve this?

#### Additional Context:
I have searched through the existing code and documentation but couldn't find a clear reference to adding frequency data. Any guidance or suggestions would be highly appreciated. Thank you!

rt-bishop, 2024-09-22:
Hey @5Breeze, sorry for such a late reply. I took some time away from the project and wasn't following the issues.

Basically the most interesting part for you would be the connection between RadarViewModel and the BTReporter. [Here](https://github.com/rt-bishop/Look4Sat/blob/main/app/src/main/java/com/rtbishop/look4sat/presentation/radarScreen/RadarViewModel.kt#L60) you can find the LiveData which stores a list of transponsers for the selected satellite. Then you'd need to come up with the way to select which transponder's data to send to the BTReporter. For example by clicking the transponder recycler view item or just by selecting a particular transponder index from the list.

Then a good example to look at would be the commented out function [here](https://github.com/rt-bishop/Look4Sat/blob/main/base/src/main/java/com/rtbishop/look4sat/utility/DataReporter.kt#L36). This is essentially all you need to actually send the data, just pass your transponder data as a parameter and make the structure similar to reportRotation function from the BTReporter. The full list of commands that rotctl devices can send/receive is [here](https://manpages.ubuntu.com/manpages/trusty/man1/rotctl.1.html#commands). Hope that clarifies some bits.

5Breeze, 2024-09-22:
Thank you very much for answering my question. I am not using a rotctl device, but rather trying to use a custom Bluetooth serial module (which is connected to a mobile radio) to transmit satellite frequency information to a Quansheng K5 radio. The goal is for the device to adjust the frequency in real-time based on the satellite transceiver selected in the phone. I am aware that some devices can already utilize a fully offline Doppler shift feature, such as the GD77 firmware, but I would still like to integrate Look4Sat to achieve real-time frequency adjustments. I appreciate you pointing me to the relevant code, and I will study it carefully. Thanks again!