rt-bishop/Look4Sat #146: [FEATURE REQ] Build in audio recorder

michalonex, 2024-10-09:
When making contacts it's basically nesesery to use audio recorder. Maybe building it in to the app with timestamp of AOS and LOS, sat name, date and AOS in title is an idea for you.

It can by primary recorder but also secondary for backup for Sat Hams.

3geETR, 2026-09-26:
That would be really great. It’s been two years and there’s been no activity on the issue, but it definitely needs to be added. Right now, I’m opening my phone’s voice recorder and going back to Look4Sat for doppler.

rt-bishop, 2026-09-27:
Thanks for bumping this, and I can definitely understand the use case - having the recording tied to the satellite pass would be pretty convenient. That said, I don't currently plan to implement this feature myself. There are quite a few other things I'd rather spend my limited development time on.

That doesn't mean I'm opposed to the idea, though. If someone in the community finds it useful and would like to take it on, I'd be happy to consider a PR. These days there are also plenty of AI tools that can make contributing to this project much easier, even if you're not already familiar with the codebase.

So if this is something you'd really like to see in Look4Sat, a community contribution would probably be the best way forward =)

3geETR, 2026-09-27:
I see, thank you. I can't say I have a lot of free time either, but if I find the time, I'll do it and open a PR. Do you have an roadmap for now?

rt-bishop, 2026-09-28:
This app's development was always quite ad-hoc so there never was a roadmap. However in the short term my focus would be changing the app's navigation to allow for more screens. Currently screens seem a bit too dense and easily confuse/overload users. I think that replacing a BottomNav with a NavigationDrawer would make the app more straightforward. The drawer could store info and links to the rest of the app.

Current screens can't just keep taking new features in. This seems to be a good time to review the placement of certain UI elements and the screens interaction logic. Also I hope to create some in-app documentation and/or onboarding to make the app less confusing.