rt-bishop/Look4Sat #152: Sending Up/Downlink-Frequencies to Transceiver via Bluetooth

Beiri22, 2025-02-14:
Hello,

I have recently checked out your repo / dev branch and (quick and dirty) implemented a mechanism for sending the up-and-downlink frequncies to my icom ic 705 transceiver via BT. Its far from perfect, some things are hard coded; i'm not a kotlin programmer; but are you interested in seeing it, maybe using it as basis for officiel support?

let me know.

rt-bishop, 2025-02-14:
Hey @Beiri22, that sounds amazing =) I was planning to get there at some point, but struggled to find time. Of course, if you could open a PR, I can have a proper look/refactor before merging! How do you find the dev branch build in general? Is it better than the Live version? I'm hoping to finish it in the next few months or so.

Beiri22, 2025-02-14:
pls see #153

rt-bishop, 2025-02-22:
Thanks a lot for the feedback! I'm happy to be on the right track. Just wish I had more free time, ha-ha =)

I had a look at the PR and was quite impressed with the amount of changes, especially not from a kotlin dev. I'm going to merge your PR as is and obviously clean things up a bit afterwards. Thanks a lot for this contribution, it makes me really happy to know that people find this app useful and even take it a step further to send the code!

Beiri22, 2025-02-22:
In germany we say "Gut Ding will Weile haben" -- something like "Good things need their time"

:-) I feel appreciated. Programming languages do use similar concepts and chatgpt helps finding possibly usable constructs and describing them, and motivation helps :-) Maybe the hardware-specifics can be put in a separate class, so that adding new hardware is easy to manage. I hope we can find a way to get the tone settings to the device, too. Do you have hardware to test on?

Beiri22, 2025-02-22:
And: Do not forget that in app/build.gradle.kts the applicationId is changed to "com.rtbishop.look4satic705" -- this was only to have both versions next to each other on my phone. You might revert that!

rt-bishop, 2025-02-22:
Thanks! No, unfortunately, I don't have any hardware to test things on. ICOMs in general are quite expensive =) I'll try to add minimal changes to BluetoothCIV, and will also look into HAMLIB and RIGCTL which may help me come up with an abstracted solution.