rt-bishop/Look4Sat #155: Unable to complete filtering

foy1002, 2025-04-26:
In the filter box, enter the satellite you want to search for, such as ISS, and the page will pop up. After checking, click the confirm button below. The homepage cannot display satellite data based on the filtering results. If you enter the filtering interface again, an error will occur. After entering the filtering box, the filtering will become invalid.
VER:3.1.4
system:ANDROID 15

DimaGepard, 2025-04-26:
I've got the same problem with filtering - main page shows all the sats and ignores the filter

rt-bishop, 2025-05-03:
Hey there! Sorry for such a late response. It's hard to create one size fits all solution for multi selection for 9000 satellites in the app with overlapping categories, transceivers and the rest. When you add a single satellite like ISS to the selection it doesn't unselect the rest of the satellites that you selected previously. My suggestion would be to go to the selection page, then click on the satellite icon in top right and unselect everything there. Then click on "Selected type" and make sure All category is selected. Then click unselect all in the bottom and then select only the satellites you need. 

After you click the yellow confirm button, you'll be taken to passes page which will show you the passes for all your selected satellites that are ordered by their time of AOS (Acquisition of Signal/Sight). Some satellites pass overhead only a few times a day, so you'll see them in the bottom of the list. Additional filtering is happening under the "filter" button on the passes screen. In that filter dialog you can configure for how long to make the predictions. Let's say you set just 8 hours ahead, but the satellite you're looking for will only come in 9 hours, so you won't see it in the list of passes. I'd recommend setting at least 24 hours or so.

It may not be super straightforward, but I'm hoping this info clarifies the filtering system quite a bit.

closed 2025-10-18 by rt-bishop