rt-bishop/Look4Sat #156: Unable to remove satellite

FastThenLeft, 2025-04-30:
Version 3.1.4

I had "UVSQ-SAT" in my list of satellites. I didn't use the program for months. Now, "UVSQ-SAT" is no longer in the TLEs so there is no way to delete it from showing up on the main page!

Thanks for such an awesome app!

-David

FastThenLeft, 2025-04-30:
For anyone following this, you can go in and clear the data and cache for the app and then redo your app settings. That successfully got rid of the satellite that was no longer in the TLEs.

rt-bishop, 2025-05-03:
Hey there! Sorry for such a late response. Indeed the manual clear of the database is the intended way of handling this use case. Some users rely heavily on manual data import and that's why I can't just reset the database with every single update and wipe their data. That's why I just overwrite the orbital objects based on their NORAD CatID and satellites after deorbiting remain in the database. 

Manual cleanup is definitely a solution, however I don't see why just unselecting them via selection page doesn't help. Make sure the "All" category is selected and you haven't additionally filtered the main list by transceivers via the satellite icon in the top right corner of the selection page.

closed 2025-10-18 by rt-bishop