rt-bishop/Look4Sat #160: Tevek2 are missing

petrkr, 2025-06-01:
Add this data sourcec which contains also new Tevel2 satellites

http://www.amsat.org/amsat/ftp/keps/current/nasabare.txt

I added it manually by file, but on next online update they disappear

petrkr, 2025-06-09:
Do not know... If I will check that txt file, its name is right TEVEL2-x

```
petrkr@archlinux /t/a> grep -i tevel nasabare.txt
TEVEL2-4
TEVEL2-5
TEVEL2-6
TEVEL2-1
TEVEL2-2
TEVEL2-3
TEVEL2-8
TEVEL2-9
TEVEL2-7
```

if I will try to find it by other name (like callsign or something) it is also not there.

But strange is, if I will download this .txt file on phone and update it by using "File" option instead "Online", then they appears.

I also tried clear the database. I have version 3.1.4 if that is latest

Also strange thing is. If I will use online, I will get 12060 satellites, if I will after it use File update, only two more satellites appear (12062), but all tevels (like 9 of them) are visible since that.

petrkr, 2025-06-11:
Thanks for review, there is problem with names that I really do not know where I have to find such names, like how to know tevel-2 is some magic number. I tried search some callsign, but problem of gooogle and tevel2-x sats is whenever you will google tevel2-2 you will get hundreds informations about 2 years old tevel-2 as AI at google thing you did typo.

Do understand this issue, do not know what is better. 

Best of thay would be aliases, but that is something what is not doable in free time.

I also have some projects which i do at freetime, so i understand u.

rt-bishop, 2025-07-26:
I'd suggest clearing the app's storage and cache via Long click - App info. And then to do a satellite data update. These steps have solved it for users with similar issue.