rt-bishop/Look4Sat #162: Online Update

ArminXXXXX, 2025-07-26:
online update is not working on s24 ultra, Android 15. i am looking for a database now. This is not consumer friendly

rt-bishop, 2025-07-26:
Well, this app doesn't try to be friendly, it's my side project. It even says in the app "comes as is, with no warranty" =)

On a serious note, I've verified the online update on 5 various devices and everything works as expected. I've seen this happening for some users in sanctioned countries, so it may be a blocking/VPN or data provider server fault issue, which is out of the app's scope.

Also the app performs the data update in parallel, so even if it shows "Update completed with errors", the successful fetches would still stay in the database. Alternatively, I'd suggest to manually fetch https://celestrak.org/NORAD/elements/gp.php?GROUP=active&FORMAT=tle, save it to your device and then perform the manual update from File. Hope this helps.

ArminXXXXX, 2025-07-27:
thanks, it was  my vpn. will be good to use an error massage in you app 🤭