rt-bishop/Look4Sat #167: 在中国大部分地区无法更新转发器 In most regions of China, it is not possible to update transceiver.

Mubi-Baihua, 2025-08-29:
### 因为网络的原因在中国大部分地区无法更新转发器。
### Due to network reasons, the transceiver cannot be updated in most regions of China.

**我有一下几点建议：**
**I have the following suggestions:**

1. 添加自定义数据源功能。
     Add the function of adding custom data sources.

2. 在追踪卫星时，让使用者手动输入转发器信息。
    When tracking satellites, users can manually input the transceiver information.

3. 在自动更新卫星信息时，检查卫星信息源是否可用。如果不可用就不进行自动更新。
    When updating satellite information automatically, check whether the satellite information source is available. If it is not available, do not proceed with the automatic update.

> 希望你能采纳这些建议！Look4Sat对于我们来说真的很棒！😀
I hope you can adopt these suggestions! Look4Sat is really great for us！😀

[image]

rt-bishop, 2025-09-02:
Hey there! Thanks for raising this issue, it seems to be a tricky one.

The app only accepts transceivers data from SatNOGS project's [API](https://db.satnogs.org/api/transmitters/?format=json&status=active). If this link doesn't open for you via browser and the app says "Updated with erros", it actually skips the unavailable info and updates the rest of the information like TLE.

I've been thinking to also let the user manually import the transceivers JSON if they can obtain it via other means. However it has to be in the exact same format as the one from SatNOGS. If you have any other transceivers data provider, please let me know.

Also, in the future I'm planning to integrate the ability to add transceivers manually via the Radar page. Hope that info helps.

rt-bishop, 2026-02-08:
Indeed, that PR should definitely solve the problem!
Version 4.0.2 is currently in review and should be available within 24 hours.